Hypnotriod/midi-box-stm32 #5: MIDI_IN_PORTS_NUM undeclared

kisielk, 2023-03-18:
I followed the README and put `MIDI_IN_PORTS_NUM` and `MIDI_OUT_PORTS_NUM` into `main.h` but I get an error when compiling `usbd_midi.c`:

```
arm-none-eabi-gcc "../Middlewares/ST/STM32_USB_Device_Library/Class/MIDI/Src/usbd_midi.c" -mcpu=cortex-m0plus -std=gnu11 -g3 -DDEBUG -DUSE_HAL_DRIVER -DSTM32G0B1xx -c -I../Core/Inc -I../Drivers/STM32G0xx_HAL_Driver/Inc -I../Drivers/STM32G0xx_HAL_Driver/Inc/Legacy -I../Drivers/CMSIS/Device/ST/STM32G0xx/Include -I../Drivers/CMSIS/Include -I../USB_Device/App -I../USB_Device/Target -I../Middlewares/ST/STM32_USB_Device_Library/Core/Inc -I../Middlewares/ST/STM32_USB_Device_Library/Class/HID/Inc -I"/Users/kamil/STM32CubeIDE/workspace_1.12.0/Abraxas/Middlewares/ST/STM32_USB_Device_Library/Class/MIDI/Inc" -O0 -ffunction-sections -fdata-sections -Wall -fstack-usage -fcyclomatic-complexity -MMD -MP -MF"Middlewares/ST/STM32_USB_Device_Library/Class/MIDI/Src/usbd_midi.d" -MT"Middlewares/ST/STM32_USB_Device_Library/Class/MIDI/Src/usbd_midi.o" --specs=nano.specs -mfloat-abi=soft -mthumb -o "Middlewares/ST/STM32_USB_Device_Library/Class/MIDI/Src/usbd_midi.o"
In file included from ../Middlewares/ST/STM32_USB_Device_Library/Class/MIDI/Src/usbd_midi.c:43:
/Users/kamil/STM32CubeIDE/workspace_1.12.0/Abraxas/Middlewares/ST/STM32_USB_Device_Library/Class/MIDI/Inc/usbd_midi.h:49:41: error: 'MIDI_IN_PORTS_NUM' undeclared here (not in a function)
   49 | #define USB_MIDI_REPORT_DESC_SIZE      (MIDI_IN_PORTS_NUM * 16 + MIDI_OUT_PORTS_NUM * 16 + 33)
      |                                         ^~~~~~~~~~~~~~~~~
/Users/kamil/STM32CubeIDE/workspace_1.12.0/Abraxas/Middlewares/ST/STM32_USB_Device_Library/Class/MIDI/Inc/usbd_midi.h:50:41: note: in expansion of macro 'USB_MIDI_REPORT_DESC_SIZE'
   50 | #define USB_MIDI_CONFIG_DESC_SIZE      (USB_MIDI_REPORT_DESC_SIZE + USB_MIDI_CLASS_DESC_SHIFT)
      |                                         ^~~~~~~~~~~~~~~~~~~~~~~~~
../Middlewares/ST/STM32_USB_Device_Library/Class/MIDI/Src/usbd_midi.c:127:48: note: in expansion of macro 'USB_MIDI_CONFIG_DESC_SIZE'
  127 | __ALIGN_BEGIN static uint8_t USBD_MIDI_CfgDesc[USB_MIDI_CONFIG_DESC_SIZE]  __ALIGN_END =
      |                                                ^~~~~~~~~~~~~~~~~~~~~~~~~
/Users/kamil/STM32CubeIDE/workspace_1.12.0/Abraxas/Middlewares/ST/STM32_USB_Device_Library/Class/MIDI/Inc/usbd_midi.h:49:66: error: 'MIDI_OUT_PORTS_NUM' undeclared here (not in a function)
   49 | #define USB_MIDI_REPORT_DESC_SIZE      (MIDI_IN_PORTS_NUM * 16 + MIDI_OUT_PORTS_NUM * 16 + 33)
      |                                                                  ^~~~~~~~~~~~~~~~~~
/Users/kamil/STM32CubeIDE/workspace_1.12.0/Abraxas/Middlewares/ST/STM32_USB_Device_Library/Class/MIDI/Inc/usbd_midi.h:50:41: note: in expansion of macro 'USB_MIDI_REPORT_DESC_SIZE'
   50 | #define USB_MIDI_CONFIG_DESC_SIZE      (USB_MIDI_REPORT_DESC_SIZE + USB_MIDI_CLASS_DESC_SHIFT)
      |                                         ^~~~~~~~~~~~~~~~~~~~~~~~~
../Middlewares/ST/STM32_USB_Device_Library/Class/MIDI/Src/usbd_midi.c:127:48: note: in expansion of macro 'USB_MIDI_CONFIG_DESC_SIZE'
  127 | __ALIGN_BEGIN static uint8_t USBD_MIDI_CfgDesc[USB_MIDI_CONFIG_DESC_SIZE]  __ALIGN_END =
```

I notice that file does not `#include "main.h"` anywhere so I'm not sure how it's supposed to pick up the number of ports based on your scheme?

Hypnotriod, 2023-03-18:
I guess I forgot to mention that you need to include 
`usb_device.h` in main file as well. Could you recheck?

Hypnotriod, 2023-03-18:
Ok. It is actually not *my* scheme, but ST guys. They did things as they did.  
You can follow the includes route starting from my `usbd_midi.c` file:  
`usbd_midi.c` -> `usbd_midi.h` -> `usbd_ioreq.h` -> `usbd_def.h` -> `usbd_conf.h` -> `main.h`  
Perhaps you are using some old ST libraries, that's why this issue occurs. But you can always put `MIDI_IN_PORTS_NUM` and `MIDI_OUT_PORTS_NUM` in `usbd_desc.h` for example.

kisielk, 2023-03-18:
I'm using the latest HAL for STM32G0. I went down the chain and it seems that `usbd_conf.h` doesn't include `main.h` in a freshly-generated project. Maybe it would be better to suggest those defines go in there?

Hypnotriod, 2023-03-18:
Actually I like more when all macros are located in one place in main.h  
Anyway I've added alternative path for those defines below in the README as well

kisielk, 2023-03-18:
Sure. I'm gonna leave them in `usbd_conf.h` since that's where all the other USB-related macros are, seems to make sense rather than having the USB stack depend on the application main.

Hypnotriod, 2023-03-18:
Yes, this will be the best place to have them (logically). But `usbd_conf.h` doesn't seem to have any "user sections". So, when you generate new code with CubeMX, you may lose these macros.

kisielk, 2023-03-18:
In mine it has this section:
```
/* Private variables ---------------------------------------------------------*/
/* USER CODE BEGIN PV */
#define MIDI_IN_PORTS_NUM   0x01 // Specify input ports number of your device
#define MIDI_OUT_PORTS_NUM  0x01 // Specify output ports number of your device
/* USER CODE END PV */
```

Hypnotriod, 2023-03-18:
Interesting...)

Hypnotriod, 2023-03-18:
Ok. I'll mention this file instead.

closed 2023-03-19 by Hypnotriod