Hypnotriod/midi-box-stm32 #7: Detected as a MIDI but cannot send any message to the computer.

nt2ds, 2023-09-06:
I used your library and managed to make the STM32 be detected from the computer but cannot get it to send any actual message.
Can you provide an example of how I should do it? I read the documentation of USB MIDI many times, but I get lost after it starts talking about the MIDI Jacks.

timrosskamp, 2023-09-06:
```c
uint8_t usb_midi_report[4] = {
    0x19, // cable number, code index number
    0x90, // note on, channel 1
    0x3C, // note number
    0x7F, // velocity
};

USBD_MIDI_SendReport(&hUsbDeviceFS, usb_midi_report, 4);
```

this is a simple example of sending a note-on event that worked for me. maybe will help you.

nt2ds, 2023-09-06:
Nop, still not working, I also asked on reddit about the problem and neither the solution someone gave me there worked, so the problem must be starting somewhere else.

nt2ds, 2023-09-06:
Can we maybe get on a PM on something like discord to help me resolve it and do it faster? We can post the solution here after, GitHub Issues aren't the best way.

Hypnotriod, 2023-09-06:
Are you checking the midi device status for idle before sending the message?

nt2ds, 2023-09-07:
I use USBPcap to see if I am getting any message, and yes I get. But no MIDI app can see them (of course I am trying one app open at a time). And yes, I use the check. I will post screenshot.
[image]
[image]
[image]

nt2ds, 2023-09-07:
Also to mention, I use the NUCLEO-F439ZI board. And the "USB -> Enable FS" Option does not exist, I just choose USB device and set it to HID and USB OTG FS is set to DEVICE_ONLY.
Also, my usbd_conf.c file does not have this line of code: " HAL_PCDEx_PMAConfig((PCD_HandleTypeDef*)pdev->pData , 0x81 , PCD_SNG_BUF, 0x100); " Thus, I only change the include and this "static uint32_t mem[(sizeof(USBD_HID_HandleTypeDef)/4)+1];" with this " static uint32_t mem[(sizeof(USBD_MIDI_HandleTypeDef)/4)+1]; ".
Every other MIDI device (BCF2000, APC Mini, korgNANO kontrol2) are working normally. Is there a problem with this specific board?

Hypnotriod, 2023-09-07:
If you defined only one `MIDI_IN_PORTS_NUM`, so try to use cable 0 instead of 1 to not confuse the app:
```C
uint8_t usb_midi_report[4] = {
    0x09, // cable number, code index number
    0x90, // note on, channel 1
    0x3C, // note number
    0x7F, // velocity
};
```

nt2ds, 2023-09-07:
Yesss, it finally worked. Cannot thank you enough!!!

closed 2023-09-07 by Hypnotriod