Hypnotriod/midi-box-stm32 #8: Trouble with receiving midi event from host

timrosskamp, 2023-09-06:
Hi, first of all thank you very much for this stm32 midi project!

I'm trying to implement this library on an STM32L1 and so far I've been able to send midi events from the STM to my computer (macos), I can monitor the event using a MIDI Monitor App. But receving midi messages is not working, the STM is only detected as a midi source and not a destination. Could there be any issue with the usb descriptor? I've looked trough it, but I don't know enough about the USB protocol to troubleshoot the code.

Not sure this is helping, but I also have an arduino micro on hand where I implemented the USBMIDI library, which is detected as an input and output midi device. Although the code is hard to compare, since the framework is completely different.

I hope you can help me or even just give me some directions on how to approach to debug this.

Hypnotriod, 2023-09-06:
Have you specified the `MIDI_IN_PORTS_NUM`?
And how do you receive the message?

timrosskamp, 2023-09-06:
Yes, I did define those:

```c
#define MIDI_IN_PORTS_NUM   0x01 // Specify input ports number of your device
#define MIDI_OUT_PORTS_NUM  0x01 // Specify output ports number of your device
```

This is the code would handle receiving the midi messages, but I can't even send midi messages from my computer to the STM32, because it's not recognized as a midi destination and only a source. I've tried it in Ableton and in MIDI Monitor and both Apps only list it as MIDI source.

```c
// main.c
#define MIDI_BUFFER_LENGTH           256

static uint8_t buffUsb[MIDI_BUFFER_LENGTH] = {0};
volatile static uint8_t buffUsbNextIndex = 0;

void USBD_MIDI_DataInHandler(uint8_t *usb_rx_buffer, uint8_t usb_rx_buffer_length)
{
  while (usb_rx_buffer_length && *usb_rx_buffer != 0x00)
  {
    buffUsb[buffUsbNextIndex++] = *usb_rx_buffer++;
    buffUsb[buffUsbNextIndex++] = *usb_rx_buffer++;
    buffUsb[buffUsbNextIndex++] = *usb_rx_buffer++;
    buffUsb[buffUsbNextIndex++] = *usb_rx_buffer++;

    usb_rx_buffer_length -= 4;
  }
}
```

Here you can see the Device in MIDI Monitor, note that it's only listed under MIDI sources:

[image]

For reference, this is my Arduino Micro which is listed as source and destination.

[image]

Hypnotriod, 2023-09-07:
Try to change `USBD_PID_FS` in `usbd_desc.c`. Maybe it was cached, or something. Have no other idea for now.

Hypnotriod, 2023-09-11:
Have you managed your issue?

closed 2023-09-13 by Hypnotriod

timrosskamp, 2023-09-18:
I'm still working on it. If I figure it out, I'll let you know.

timrosskamp, 2023-10-13:
Ok. So I had some time to work on this and I managed to get this working for Windows. I can send MIDI Events using MIDI Tools on Windows and the Device is receiving the data fine, but the problem remains with my Mac. I've installed the same Software "MIDI Tools" on my Mac and there I don't have this Device as a Midi Output available, it's only listed as an Input Device. This makes sending MIDI Events from the Host (Mac) to the Device not possible.

The only comparision I have is an Arduino Micro with a different library for MIDI which can send and receive MIDI with Mac  and I used Wireshark on WIndows to compare the USB interface descriptors and I could not find a meaningful difference between them. So I don't think the USB descriptors are the problem.

Any ideas? I'm honestly not sure what else to debug.

Hypnotriod, 2023-10-14:
Have you tried with last version of midi middleware I made? Have you tried to change PID value?

Quotation, 2025-01-28:
Mac OS caches MIDI device information. Just delete the MIDI device from `Audio MIDI Setup - MIDI Studio`, then connect the device again to refresh. This solves my similar issue.