Hypnotriod/midi-box-stm32 #11: Endpoint inconsistency

x-moe-x, 2024-07-24:
Hi,

thanks for the great work!

I found an inconsistency regarding endpoint configuration: In the descriptor the EPs are configured as BULK, in USBD_MIDI_Init however they are opened as Type Interrupt.

Kind regards,

Moe

Hypnotriod, 2024-07-24:
I see. This could be the reason of that [Double Buffering](https://github.com/Hypnotriod/midi-box-stm32/issues/9) issue...
Could you please confirm that replacing `USBD_EP_TYPE_INTR` with `USBD_EP_TYPE_BULK` in `USBD_MIDI_Init` -> `USBD_LL_OpenEP` calls does not brake anything?

Hypnotriod, 2024-07-24:
Anyway. Have checked it on the device and seems no issues so far. Did the [commit](https://github.com/Hypnotriod/midi-box-stm32/commit/231c11da816e36df618e5c224df6ebc4f2f5ddc7)

closed 2024-07-24 by Hypnotriod