Hypnotriod/midi-box-stm32 #14: Sending data to host on second port

LispEngineer, 2025-05-13:
Hi there. Thanks for this! I have it working on the Nucleo F767ZI board. It is published here on GitHub.

I configured two IN ports (STM to Windows host). Your code did not include any ability to specify which IN to send on, so I made an extended version. However, so far I have been unable to get it to send on the second port, even though MidiView shows both ports. How is that done, please? Thanks!

Hypnotriod, 2025-05-13:
Hi. From the [README.md](https://github.com/Hypnotriod/midi-box-stm32/blob/master/README.md):
"The `port` means the `cable` number or `MIDI Jack` associated with the endpoint that is transferring the data."
"Midi event packet structure:"
```txt
|  4 bits  |  4 bits  |  4 bits  |  4 bits  |  8 bits  |  8 bits  |
|          |          |        MIDI_0       |  MIDI_1  |  MIDI_2  |
|   Cable  |   Code   |  Message |  Channel |  Byte 1  |  Byte 2  |
```
"Example sending one event packet to the host device:"
```C
extern USBD_HandleTypeDef hUsbDeviceFS;
uint8_t reportBuffer[4] = {
  // cable - represents physical/virtual port number (0 - 15) of the device
  // code - in general cases is equal to midi message
  (cable << 4) | code,
  (message << 4) | channel,
  messageByte1,
  messageByte2,
};
...
while (USBD_MIDI_GetState(&hUsbDeviceFS) != MIDI_IDLE) {};
USBD_MIDI_SendReport(&hUsbDeviceFS, reportBuffer, 4);
```
P.S.: I used the `port` term there to do not interfere with the `MIDI Jack` term. Because in the usb midi specification the `IN` direction means toward the `host`. But in my case the `MIDI_IN_PORTS_NUM` macro means the number of midi inputs on your device.

Hypnotriod, 2025-05-13:
Have fixed the `USBD_MIDI_DataInHandler` callback with the [commit/077d5732952054e7df90ac934a607a5a22f22e41](https://github.com/Hypnotriod/midi-box-stm32/commit/077d5732952054e7df90ac934a607a5a22f22e41) 
Now the `USBD_MIDI_DataInHandler` will receive the correct length of the report instead of previously hardcoded `MIDI_EPOUT_SIZE`

Hypnotriod, 2025-05-14:
[image]
[image]
The `cable` is your `port` number! Because it is associated with the `cable`!
Sorry but:))
> Three shall be the number thou shalt count, and the number of the counting shall be three. Four shalt thou not count, neither count thou two, excepting that thou then proceed to three. Five is right out.

* The `note off` report on port `1`:
```c
// Cable = 0 hex, Code = 9 hex (Note On), Message = 9 hex (Note On), Channel = 0 hex, Note = 3C hex (Middle C), Velocity = 7F hex (max)
[0x08, 0x80, 0x3C, 0x7F]
```
* The `note off` report on port `2`:
```c
// Cable = 1 hex, Code = 9 hex (Note On), Message = 9 hex (Note On), Channel = 0 hex, Note = 3C hex (Middle C), Velocity = 7F hex (max)
[0x18, 0x80, 0x3C, 0x7F]
```
And if it does not show up with `2 midi in` interfaces after you change the `MIDI_IN_PORTS_NUM` to 2 - try to reinstall the device in your OS. There were issues that OS "caches" the descriptor of the device. Or try to change the `USBD_PID_FS` in the `usbd_desc.c`

Hypnotriod, 2025-05-15:
> You're saying that regardless of the number of MIDI_IN_PORTS_NUM defined, USB MIDI uses only one actual IN port.  

It is using one `IN` usb endpoint...

Hypnotriod, 2025-05-16:
Whatever... Did the code and documentation refactoring once again.
No "obscure" `IN` / `OUT` usb endpoint references.
So the function names are now: `USBD_MIDI_SendPackets`, `USBD_MIDI_OnPacketsReceived`, `USBD_MIDI_OnPacketsSent`.
The `report` term was related to the HID class. So I changed it to the `data`.

closed 2025-05-16 by Hypnotriod