Hypnotriod/midi-box-stm32 #16: I would like to run this middleware on the STM32F407, can you support me?

tunaingot, 2025-05-17:
I edited the source code while looking at README.md and got to the point where it was detected as a USB-MIDI device by macOS.

```
/* USER CODE BEGIN EndPoint_Configuration_HID */ // this section may be absent if your device has no PMA, so no action required
```
I don't know what PMA is, but I couldn't find this description in usbd_conf.c.
I would like to try receiving USB-MIDI first.

```
USBD_MIDI_GetState(&hUsbDeviceFS) == MIDI_IDLE
```
This status does not become MIDI_IDLE.

```
void USBD_MIDI_OnPacketsReceived(uint8_t *data, uint8_t len)
```

I override this function, but this function is not called.
Where is the first definition of this function?

Will you accept this kind of support?
I want to create a USB-
MIDI interface using this middleware.
And I would like to spread it in Japan so that it can be used with the STM32F.

[image]

Hypnotriod, 2025-05-17:
Hi. It depends on how you are checking the `MIDI_IDLE` flag.
Could you show your code snippet?

Hypnotriod, 2025-05-17:
And if it happens that you have the `STM32F407DISCOVERY` board, which I have - you can try to run this `STM32CubeIDE` test project I made:
[midi-test-stm32f407-discovery.zip](https://github.com/user-attachments/files/20263877/midi-test-stm32f407-discovery.zip)
I tested it on that bloody MAC as well. Open the [studiocode.dev/webmidi-tester](https://studiocode.dev/webmidi-tester/) in Chrome (just don't mess up the permissions) and send `90 00 7F  90 01 7F  90 02 7F  90 03 7F` to light up all the 4 LEDs. Also it sends the note on and note off messages each ~second.

tunaingot, 2025-05-17:
Thank you for your prompt advice!

I had confirmed the operation with ```STM32F407Discovery```.
I have confirmed that it works with my macOS MIDI app.

The Disovery test project you sent me worked fine.
It is as shown in the attached photo.

Comparing the source of the test project you sent me and the project I downloaded from GitHub, I found that the contents of ```usbd_midi.c``` and ```usbd_midi.h``` are different.

I will first use the project you sent me.

Thanks for the quick response!

[image]

[image]

Hypnotriod, 2025-05-18:
Ah. I see it now... I did the "refactoring" recently.  
So I renamed the `USBD_MIDI_DataInHandler` to the `USBD_MIDI_OnPacketsReceived`. Also the `USBD_MIDI_SendReport` to the `USBD_MIDI_SendPackets`. To not to interfere with the `IN` and `OUT` terms related to the USB endpoints. Also `report` term is related to the HID class...

closed 2025-05-18 by Hypnotriod